rosmod/webgme-rosmod #26: Deadlines for component instances

pranav-srinivas-kumar, 2016-03-29:
We want to be able to set deadlines for component instances so that we don't have to recompile during experimentation.

finger563, 2016-04-11:
Will need to figure out how this will work in the META since the component instance only has indirect references to the original component and its timers/ports and their attributes.

finger563, 2016-04-12:
We may want to change to not have `Component Instance` be a meta-type; if we allow components in the deployment model; we can say that the user should create the component by dragging it in from the tree browser and selecting `instance`.  This would allow properties to be inherited and also overridden.

finger563, 2016-04-14:
With respect to creating an `instance` of a `component`, we'd have to figure out how to determine the original component's name, since the user can rename the component instance we'd lose track of the name of the component library required for it.  Additionally there are some deployment specific concerns that would need to be moved from the current `Component Instance` to the `Component` (e.g. `SchedulingScheme`, `EnableLogging`, and `LogingUnit`).  Probably need to discuss this further.

finger563, 2016-04-25:
The only missing dependency now is the resolution of https://github.com/rosmod/rosmod-actor/issues/3 before we can close this issue.

commit 7f10e227 (2016-04-25): updated to properly use JSON config file; uses new rosmod-actor code; now port properties (priority, deadline, period) are all handled by deployment instead of code compilation.  closes #26
closed 2016-04-25 by finger563