rosmod/webgme-rosmod #29: We need a Pre-Experiment script launcher

pranav-srinivas-kumar, 2016-03-30:
Currently, in the Run_Experiment plugin, we have a hard-coded command "'export DISPLAY=:0.0'" in the start_processes function. Such commands are a property of the application and not the deployment infrastructure. So, commands like these need to be prepared in the model and therefore required as a concept in the metamodel.

finger563, 2016-04-26:
From our discussion, this seems to be a property of the component itself (e.g. because the specific component requires the use of a display?) and therefore a multi-line string attribute akin to a bash script should be provided for pre-execution commands?  The user would not need to type things like `export LD_LIBRARY_PATH=$PWD:$LD_LIBRARY_PATH` or `export ROS_IP=<IP>`, since those are still handled by the deployment infrastructure, but we'd need a way to input commands after that which are required by the user.

finger563, 2017-04-17:
Multi-line attributes `startup` and `teardown` might be useful.  The main issue is how to delineate commands that might be different for different systems.  Initially, not allowing system-specific info but just enabling some more customization of components / component instances might be beneficial.

commit 9bf20263 (2018-07-29): Merge pull request #274 from rosmod/issue-269

Add Script Node meta type Closes #269 and closes #29
closed 2018-07-29 by finger563