rosmod/webgme-rosmod #42: Need a way to specify in the metamodel which user to deploy an experiment as

pranav-srinivas-kumar, 2016-04-04:
We currently have no way to specify in the model which user to deplpy an experiment as -- ROSMOD always chooses the first available user.

finger563, 2016-04-11:
Difficult w.r.t. how the user should specify this information: plugin doesn't have access to data during config time (and might be cumbersome) and the experiment only has pointers to the deployment and the system, the map between the two is generated on the fly (and is where the user specification would need to take place).  Additionally, all processes on a host are currently mapped and run as the same user.

finger563, 2016-04-14:
Dependent on the implementation of https://github.com/webgme/webgme/issues/945

commit b29af697 (2017-11-22): closes #42 use the selected user per host. closes #138 only use hosts that have been selected in the plugin config.
closed 2017-11-22 by finger563