rosmod/webgme-rosmod #52: Experiments should have the ability to drag and drop objects onto its icon

DWatkins, 2016-04-10:
Experiments should exhibit the same click and drag behavior seen in a subscriber port object, so that a deployment or system can be dragged onto a region of the experiment icon to create a pointer to that object.

finger563, 2016-04-10:
This will be updated in a new visualizer; for now you can create an experiment by dragging a deployment or a system into the experiments aspect and it will automatically set the relevant pointer.  the other pointer will need to be set manually.

finger563, 2016-04-18:
Alternatively, this will automatically work when https://github.com/webgme/webgme/issues/954 is completed.

finger563, 2016-09-10:
SVGDecorator was updated to close webgme/webgme#954; closes this issue as well.

closed 2016-09-10 by finger563