rosmod/webgme-rosmod #54: Notifications window cannot handle large messages

DWatkins, 2016-04-10:
[image]

finger563, 2016-04-10:
Can you let me know what you were trying to do when seeing this message?

DWatkins, 2016-04-10:
Running the software generator to compile code for the copy of client-server example I was making.  Sources files generated just fine, but the error is thrown when trying to compile.  The error in its entirety is shown in the following screenshot.
[image]

finger563, 2016-04-10:
I'm going to push an update to the code which closes this specific issue (i.e. it splits the strings into multiple notifications.)  Please open a new issue with some information about the specific error you get from this compilation / copying because I believe that is another issue that needs to be taken care of.

commit c2023e5d (2016-04-10): updated code to handle large notification strings in the sfotware generator; closes #54
closed 2016-04-10 by finger563