rosmod/webgme-rosmod #106: Add a context window displaying all available objects/variables

DWatkins, 2016-05-28:
When writing component code, there is a need for a list of all objects and variables currently available to the component. Especially useful for new users or new programmers in general.

Examples:

1. When writing code within a timer, often a variable declared in the component's members section is needed.  It is a pain to click through multiple views to look up the global's name.

2. Remembering the proper namespace for a message object is not intuitive for a new user. 

3. Some default objects are not easily known without looking at the provided samples, such as a subscriber's received_data object.

At the minimum having a list for the user to reference would save time and errors. A more interactive list with the ability to either click and drag or double click an object in the list to auto-populate the proper reference at the cursor's location would be best. If that is not possible, at least copying the string to the clipboard so the user can paste it would work.

finger563, 2016-05-30:
Relevant to #69, along with some details for better implementation of code-tree browser.  We are currently working on a better UI for code editing in the WebGME window, and will likely incorporate some of these ideas for testing and feedback.

finger563, 2017-10-11:
Will be implemented as documentation added to the component configuration for webgme-codeeditor, which will be used once https://github.com/finger563/webgme-codeeditor/issues/23 is completed.

This documentation should have links to relevant external documentation (e.g. ROS Timers / API for a Timer's `Operation` attribute) as well as relevant input / output arguments and return values.  should also include documentation of useful functions and class members provided by the base ROSMOD component class.

closed 2017-11-21 by finger563