rosmod/webgme-rosmod #116: NewDefaultLayout ignores component settings

brollb, 2016-11-11:
I was importing NewDefaultLayout for another grad student to use webgme-fab and realized `NewDefaultLayout` ignores component settings for it's panels: https://github.com/rosmod/webgme-rosmod/blob/master/src/layouts/NewDefaultLayout/NewDefaultLayout.js#L27

It would be nice if it used the component settings so it was able to be used in other projects as well. If you decide against that, the config in the components.json should be removed: https://github.com/rosmod/webgme-rosmod/blob/master/config/components.json#L185-L224

finger563, 2016-11-15:
Good point; I'll update it for that and then make it its own repo.

finger563, 2017-04-05:
@brollb can you verify that c5dea6a is what you were intending?

closed 2017-04-05 by finger563

brollb, 2017-04-14:
Yep, that's what I was thinking!  Thanks!

commit c5dea6a0 (2017-04-05): updated layout to use config properly. closes #116