rosmod/webgme-rosmod #151: Find a better method of writing all logs to file when tearing down an experiment

DWatkins, 2017-07-07:
enables the user to set a high logging unit to reduce overhead and not worry about losing logs that have not be written to file.

finger563, 2017-07-07:
Short term:
* modify ROSMOD-COMM so that `rosmod_actor` has a signal handler for a non-kill / terminate signal in which it calls a function on all component instances (possibly their destructors)
* modify ROSMOD-COMM so that component base class exposes a function which encapsulates logger->write()
* modify WebGME-ROSMOD so that the stop experiment sends the new signal, waits, then sends the kill signal (currently just sends kill signal).

finger563, 2017-07-07:
Long term:
* enable communication between WebGME-ROSMOD server and ROSMOD components to allow for actor lifecycle management, reconfiguration, redeployment, etc.

finger563, 2017-10-11:
possibly can integrate with #172 which should enable sending of controls and other data to ROSMOD components at runtime.

finger563, 2017-11-21:
see: https://answers.ros.org/question/191787/sigint-handler-not-working/ and https://answers.ros.org/question/27655/what-is-the-correct-way-to-do-stuff-before-a-node-is-shutdown/ for proper ways of inserting signal handlers into nodes.

finger563, 2017-11-21:
use the `-signal` option of [pkill](https://linux.die.net/man/1/pkill)

commit 17bacb35 (2017-11-21): closes #151. StopExperiment sends SigInt to rosmod_actors. Component destructors now call write() on their logs to flush all logs.  works with the new version of rosmomd-comm which has an actor that uses a sigint handler.
closed 2017-11-21 by finger563