rosmod/webgme-rosmod #170: node.js 6.11.3 does not work

willpags1, 2017-09-26:
Fixed by just using v6.11.2

finger563, 2017-09-27:
Can you paste the error you got here?

William Emfinger, Ph.D.
Chief Technology Officer
Max Mobility, LLC.
5425 Crossings Blvd, Antioch TN

From: willpags1
Sent: Wednesday, September 27, 2017 1:03 AM
To: rosmod/webgme-rosmod
Cc: Subscribed
Subject: [rosmod/webgme-rosmod] node.js 6.11.3 does not work (#170)

Fixed by just using v6.11.2

willpags1, 2017-09-27:
I can't really without uninstalling everything and reinstalling but the error I would get when I followed the installation instructions was that I was missing a package in node.js.

finger563, 2017-10-04:
Can't replicate this issue on my end, with latest nodejs LTS. closing for now.

closed 2017-10-04 by finger563