rosmod/webgme-rosmod #172: Integrate rosbridge into webgme ROSMOD

finger563, 2017-10-11:
[rosbridge](http://wiki.ros.org/rosbridge_suite) can provide (with [rolibjs](http://wiki.ros.org/roslibjs)) run-time command and telemetry for ROS applications.  These should be integrated into webgme-rosmod to provide run-time real-time telemetry and control for running experiments.

finger563, 2017-10-16:
http://wiki.ros.org/roslibjs/Tutorials/BasicRosFunctionality covers how to interact with a rosbridge_server using the roslibjs npm module

DWatkins, 2017-10-20:
# Communication Path

Run-time real-time telemetry and control can be achieved in two steps

## Ros Interface
Achieved with rosbridge and roslibjs
1. Rosbridge being a ros package that provides a ROS<-->JSON interface
2. Roslibjs running on the same network and connected to the rosbridge server via websocket

## Telemetry and Control via OpenMCT
 a. Roslib<-->OpenMCT connection via websocket

# Component location

A rosbridge instance for every roscore. Need to avoid running rosbridge on the deployed system when possible. Potential issue with multiple rosbridge instances competing for port number.

Roslibjs and OpenMCT colocated.  

# Visualization Implementation ideas

## Telemetry configured at software generation
When generating software from model, generate a dictionary, dictionary plugin, object provider, telemetry plugin, etc. and load into openmct. Telemetry objects are then available for display composition before running experiment.  Requires heavy integration with rosmod.

## Telemetry configured at experiment run time
When a new instance of rosbridge is up, an alert is sent to roslib to establish a connection, poll rosbridge for available topics, and create telemetry objects for them.  Less integration with rosmod required.

DWatkins, 2017-11-30:
66e5c32cff8995abc476033d7368da8919214dbe

closed 2017-11-30 by DWatkins