rosmod/webgme-rosmod #182: rosbridge cannot interact with rosmod-generated packages

DWatkins, 2017-10-26:
```
[ERROR] [1509027639.165797]: Error processing request: Unable to load the manifest for package actuators. Caused by: actuators
['Traceback (most recent call last):\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 625, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/opt/ros/kinetic/lib/rosapi/rosapi_node", line 163, in get_message_details\n    typedefs = [dict_to_typedef(d) for d in objectutils.get_typedef_recursive(request.type)]\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosapi/objectutils.py", line 78, in get_typedef_recursive\n    return _get_typedefs_recursive(type, [])\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosapi/objectutils.py", line 169, in _get_typedefs_recursive\n    typedef = get_typedef(type)\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosapi/objectutils.py", line 60, in get_typedef\n    instance = ros_loader.get_message_instance(type)\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_library/internal/ros_loader.py", line 101, in get_message_instance\n    cls = get_message_class(typestring)\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_library/internal/ros_loader.py", line 88, in get_message_class\n    return _get_msg_class(typestring)\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_library/internal/ros_loader.py", line 128, in _get_msg_class\n    return _get_class(typestring, "msg", _loaded_msgs, _msgs_lock)\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_library/internal/ros_loader.py", line 163, in _get_class\n    cls = _load_class(modname, subname, classname)\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_library/internal/ros_loader.py", line 187, in _load_class\n    raise InvalidPackageException(modname, exc)\n', 'InvalidPackageException: Unable to load the manifest for package actuators. Caused by: actuators\n']
```

DWatkins, 2017-10-27:
Underlying problem with message definitions not being installed onto the system running rosbridge.  Rosbridge itself just makes use of rostopic etc.

example:

Deploy Intro To Rosmod project onto local host and open a new terminal

All topics currently being used on the network are visible.
```
 jeb@ubuntu  ~/ws  rostopic list
/ColorRGBA
/ComponentName
/rosout
/rosout_agg
```

But message contents for custom messages cannot be viewed
```
jeb@ubuntu  ~  rostopic echo /ComponentName
ERROR: Cannot load message class for [three_component_example/ComponentName]. Are your messages built?
```

If the generated src folder from the rosmod model is copied to a new workspace and modified to only contain message definitions, then built and sourced, message content can be shown.
```
jeb@ubuntu  ~/repositories/webgme-rosmod   master ●  mkdir ~/example_workspace
 jeb@ubuntu  ~/repositories/webgme-rosmod   master ●  cp -r generated/guest+samples/master/Introduction\ to\ ROSMOD/src ~/example_workspace 
 jeb@ubuntu  ~/repositories/webgme-rosmod   master ●  cd ~/example_workspace 
 jeb@ubuntu  ~/example_workspace  catkin config --extend /opt/rosmod
----------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/rosmod
Workspace:                   /home/jeb/example_workspace
----------------------------------------------------------------
Source Space:       [exists] /home/jeb/example_workspace/src
Log Space:         [missing] /home/jeb/example_workspace/logs
Build Space:       [missing] /home/jeb/example_workspace/build
Devel Space:       [missing] /home/jeb/example_workspace/devel
Install Space:      [unused] /home/jeb/example_workspace/install
DESTDIR:            [unused] None
----------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
----------------------------------------------------------------
Workspace configuration appears valid.

Initialized new catkin workspace in `/home/jeb/example_workspace`
----------------------------------------------------------------
 jeb@ubuntu  ~/example_workspace  emacs src/three_component_example/CMakeLists.txt 
 jeb@ubuntu  ~/example_workspace  rm -rf src/three_component_example/{include,src}
 jeb@ubuntu  ~/example_workspace  catkin build
----------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/rosmod
Workspace:                   /home/jeb/example_workspace
----------------------------------------------------------------
Source Space:       [exists] /home/jeb/example_workspace/src
Log Space:         [missing] /home/jeb/example_workspace/logs
Build Space:        [exists] /home/jeb/example_workspace/build
Devel Space:        [exists] /home/jeb/example_workspace/devel
Install Space:      [unused] /home/jeb/example_workspace/install
DESTDIR:            [unused] None
----------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
----------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
----------------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                                                        
[build] Updating package table.                                                                                   
Starting  >>> catkin_tools_prebuild                                                                               
Finished  <<< catkin_tools_prebuild                  [ 1.1 seconds ]                                              
Starting  >>> three_component_example                                                                             
Finished  <<< three_component_example                [ 2.0 seconds ]                                              
[build] Summary: All 2 packages succeeded!                                                                        
[build]   Ignored:   None.                                                                                        
[build]   Warnings:  None.                                                                                        
[build]   Abandoned: None.                                                                                        
[build]   Failed:    None.                                                                                        
[build] Runtime: 3.1 seconds total.                                                                               
[build] Note: Workspace packages have changed, please re-source setup files to use them.
 jeb@ubuntu  ~/example_workspace  source devel/setup.zsh
 jeb@ubuntu  ~/example_workspace  rostopic echo /ComponentName 
name: Component_2
---
name: Component_2
---
name: Component_2
---
```

DWatkins, 2017-10-27:
`CMakeLists.txt` must be modified in the package to only build msg and srv 

```
 cmake_minimum_required(VERSION 2.8.3)
project(three_component_example)

## Start Global Marker
## End Global Marker

## Check C++11 / C++0x
include(CheckCXXCompilerFlag)
CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
CHECK_CXX_COMPILER_FLAG("-std=c++0x" COMPILER_SUPPORTS_CXX0X)
if(COMPILER_SUPPORTS_CXX11)
    set(CMAKE_CXX_FLAGS "-std=c++11")
elseif(COMPILER_SUPPORTS_CXX0X)
    set(CMAKE_CXX_FLAGS "-std=c++0x")
else()
    message(FATAL_ERROR "The compiler ${CMAKE_CXX_COMPILER} has no C++11 support. Please use a different C++ compiler.")
endif()

find_package(catkin REQUIRED COMPONENTS message_generation rosmod rosmod_actor
# external packages that we depend on
std_msgs std_srvs
)

# Generate services in the 'srv' folder
add_service_files(
  FILES
  ComponentService.srv
)

# Generate messages in the 'msg' folder
add_message_files(
  FILES
  ComponentName.msg
)

## Generate actions in the 'action' folder
# add_action_files(
#   FILES
#   Action1.action
#   Action2.action
# )

# Generate added messages and services with any dependencies listed here
generate_messages(
  DEPENDENCIES
  std_msgs
)

#
## catkin specific configuration 
#
## The catkin_package macro generates cmake config files for your package
## Declare things to be passed to dependent projects
## INCLUDE_DIRS: uncomment this if you package contains header files
## LIBRARIES: libraries you create in this project that dependent projects also need
## CATKIN_DEPENDS: catkin_packages dependent projects also need
## DEPENDS: system dependencies of this project that dependent projects also need
catkin_package(
#  INCLUDE_DIRS include
#  LIBRARIES client_server_package
#  CATKIN_DEPENDS roscpp std_msgs
  CATKIN_DEPENDS message_runtime
#  DEPENDS system_lib
)

#
## Build 
#

## Specify additional locations of header files
## Your package locations should be listed before other locations
# include_directories(include)
```

finger563, 2017-10-27:
Looks good, we'll add a template for that into the `RunExperiment` plugin, which will generate and build the `msg`/`srv` for the packages similarly to the `SoftwareGenerator` plugin into the temp directory it already makes, spawn off a `rosbridge_server` process which will source the env for the output of that build before deleting the folder

might be a good idea to look into moving some of the `SoftwareGenerator` functionality into a separate file which can be used by both `SoftwareGenerator` and `RunExperiment`.

DWatkins, 2017-11-07:
a barebones solution that is more conducive to automated setup and teardown

# Required files for proper Rosbridge deserialization
* python message scripts located in `.../devel/lib/python2.7/dist-packages/[package name]
* package.xml

# Setup steps

1. Copy python scripts and package.xml to folders on host
2. update `PYTHONPATH` and `ROS_PACKAGE_PATH` env variables to point to relevant folders

closed 2017-11-07 by DWatkins