rosmod/webgme-rosmod #184: Modify software generator to store message definitions and package.xml

DWatkins, 2017-11-07:
after `catkin_make` 

* message and service definitions are located in `devel/lib/python2.7/dist-packages/[package_name]/{msg,srv}`
* package.xml manifest located in `src/[package_name]/package.xml`

need to store these files for use in deployment so all hosts have the ability to parse all topics (either through `rostopic`, `rosbridge` or some other utility

finger563, 2017-11-19:
After we store them, what should their structure look like when we need to use them again? or does it not matter if we set the PYTHONPATH and other ENV vars ourselves?

DWatkins, 2017-11-20:
As long as we append the paths to PYTHONPATH and ROS_PACKAGE_PATH, all that are needed are the deserialization python scripts and the package.xml manifest

see #182

finger563, 2017-11-20:
now generates into a `share` folder, simply spawn a shell into that folder,
```bash
echo PYTHONPATH=$PWD:$PYTHONPATH
echo ROS_PACKAGE_PATH=$PWD:$ROS_PACKAGE_PATH
```

then you'll be able to run `rostopic` and `rosbridge` with the ability to deserialize messages

finger563, 2017-11-20:
right now doing this allows `rostopic echo` to work, but rosbridge gets this failure when roslib tries to connect:
```
[ERROR] [1511199232.190217]: [Client 0] [id: subscribe:/ComponentName:1] subscribe: Unable to load the manifest for package three_component_example. Caused by: three_component_example
```

finger563, 2017-11-21:
Seems to be working now; closing.

closed 2017-11-21 by finger563