rosmod/webgme-rosmod #186: Spawn rosbridge in new shell to connect to a roscore

DWatkins, 2017-11-07:
_From @DWatkins on November 2, 2017 17:4_

Spawn a worker process to contain rosbridge and roslib interfaces for each roscore.

Rosbridge needs to run in a new shell for each roscore. 

Needs to:
1. source message definitions for all packages on the system
2. generate a unique port to reside on 

_Copied from original issue: rosmod/rosmct#2_

DWatkins, 2017-11-07:
only spawn a new shell for rosbridge to reside at.

DWatkins, 2017-11-07:
updated to be on rosmod's end.  no long need to deal with roslib interaction, which is handled in rosmct

DWatkins, 2017-11-20:
related to #182, #184, #185, #187

finger563, 2017-11-21:
Need to figure out how to properly manage rosbridge_server's lifecycle (so we can kill it when we `StopExperiment`)

DWatkins, 2017-12-06:
ocal host also needs to export ROS_IP to the ip address reachable by hosts in the system. Need for roscore to arbitrate pub/sub advertisements correctly

finger563, 2017-12-27:
Will have to update the plugin so that the plugin can know by what IP the ROSMOD system can reach the rosmod server (esp. since the server can and usually does have more than one set of IP addresses)

commit 66e5c32c (2017-11-22): closes #186 properly kill all spawned rosbridge processes.
commit 6bf9da1b (2017-12-27): closes #186 allow user to say what the IP address of the ros bridge server is.
closed 2017-12-27 by finger563