rosmod/webgme-rosmod #187: script to launch rosbridge and set up env variables with provided paths

DWatkins, 2017-11-07:
_From @DWatkins on November 7, 2017 18:51_

pass in options object 

```
{
env_variable: env_value
...
}
```

_Copied from original issue: rosmod/rosmct#7_

DWatkins, 2017-11-07:
related to #2

DWatkins, 2017-11-07:
example options
```
{
ROS_MASTER_URI: "http://localhost:11311",
PYTHONPATH: "/home/jeb/catkin_ws/devel/lib/python2.7/dist-packages",
ROS_PACKAGE_PATH: "/home/jeb/catkin_ws/src"
}
```

finger563, 2017-11-10:
this is part of #186 right?

DWatkins, 2017-11-10:
similar. I had it set up as two separate issues, this one is to create the rosbridge spawning  process, which conceivably is the creation of a new file/module, while #186 is to make changes to the run experiment plugin to utilize this new spawner.  Two issues for two separate changes to two different files.

DWatkins, 2017-12-06:
local host also needs to export `ROS_IP` to the ip address reachable by hosts in the system.  Need for roscore to arbitrate pub/sub advertisements correctly

closed 2017-12-06 by DWatkins