rosmod/webgme-rosmod #204: Constraints:  Compilation vs Compilation+Runtime

DWatkins, 2017-12-01:
Currently the same set of constraints are enforced at both compilation and runtime.  This can artificially constrain the available hosts on which to compile.  In reality there are some constraints that are only needed at runtime and have no effect on compilation.

For example, the presence of a system library is a necessary constraint for compilation, but the presence/absence of specific hardware is not pertinent until deployment.

Possible solutions:

Short term:
Add boolean to constraint (no need to update capability) to have a boolean `deployment only` attribute

Long term: 
Restructure model so that system libraries are specified with a host's definition, not within software, since in reality all that a system library does is state what is present on a particular host.

DWatkins, 2017-12-01:
Moving system libraries to a host does not address the situation of a source library being platform specific

finger563, 2017-12-01:
Long Term:
We need to update how and where we specify constraints and how we model libraries.  Currently libraries (e.g. source libraries) have no information about their requirements and so any components that require those libraries will need to have their constraints manually updated accordingly.

commit 845a8957 (2017-12-01): closes #204 Added "Deployment Only" boolean (default: true) to Constraints, only if it exists and is false can the constraint be used during compilation. updated seeds accordingly.
closed 2017-12-01 by finger563