rosmod/webgme-rosmod #205: Constraints: Disjoint Sets

DWatkins, 2017-12-01:
If hosts of the same platform have varying and conflicting constraints disjoint sets of packages for compilation can occur.

Currently, the software generator will just choose a host that can compile the largest number of packages for a specific architecture. Combine this with a disjoint set and some packages may not be compiled on all needed platforms.

DWatkins, 2017-12-01:
Current solution:

For each conflicting set of constraints within an architecture, specify a unique DeviceID to assign to the appropriate hosts so that separate compilation occurs.

DWatkins, 2017-12-01:
Distinguishing between runtime-only and software constraints will also help this situaation 
#204

finger563, 2017-12-01:
The closing of #204 should have alleviated this issue. From the best we can tell, proper system configuration and modeling should not produce disjoint sets of packages that can be compiled across hosts given the same `Architecture` and `Device ID`.  If we find that this is not convenient enough or that Disjoint sets crop up in some other way, we can re-open this issue.

closed 2017-12-01 by finger563