rosmod/webgme-rosmod #207: Show Node not functioning within hfsm

willpags1, 2017-12-04:
When you have a compilation error and click show node to take you to the error, it takes you to the component where the error occurs and duplicates the components in the code editor tree. Refreshing browser sets back to normal but it's inconvenient, makes debugging more difficult

finger563, 2017-12-04:
So this is when you click show node when the code editor is already showing?  Might be a bug in the code editor

willpags1, 2017-12-04:
Not necessarily, I think it happens when you hit show node from anywhere

finger563, 2017-12-04:
hmm i'm not able to reproduce this error, can you be more specific about the preconditions? i.e. what visualizer you have open, on what node, where did you run the plugin from, etc.

willpags1, 2017-12-04:
It occurs when trying to show results after a failed compilation. Clicking show node usually brings you to the place where the error occurred but within HFSM it brings you to the outlying component and I think if you are already in the code editor for that component it with add duplicates into tree on the left.

finger563, 2017-12-04:
hmmm I'm testing it on my local machine and it seems to work, i'll test it on the VADL server and see if I can reproduce it; maybe the two are running different versions

finger563, 2017-12-04:
was able to reproduce it; it's specifically caused when you have an error in an `External Transition`'s action, as the default visualizer for a transition is the `CodeEditor`.  It is a bug in the code editor though

commit 810f606b (2017-12-04): updated to new version of codeeditor which closes #207. also added back External transitions to code editor config for easier navigation to code errors and easier editing of their attributes.
closed 2017-12-04 by finger563