rosmod/webgme-rosmod #208: Compilation error on Raspberry Pi

willpags1, 2017-12-04:
This might be a raspberry pi error and not rosmod.

```
--------------------------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/rosmod
Workspace:                   /home/jeb/compilation/VADL+Workspace/master
--------------------------------------------------------------------------------
Source Space:       [exists] /home/jeb/compilation/VADL+Workspace/master/src
Log Space:         [missing] /home/jeb/compilation/VADL+Workspace/master/logs
Build Space:       [missing] /home/jeb/compilation/VADL+Workspace/master/build
Devel Space:       [missing] /home/jeb/compilation/VADL+Workspace/master/devel
Install Space:      [unused] /home/jeb/compilation/VADL+Workspace/master/install
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.

Initialized new catkin workspace in `/home/jeb/compilation/VADL+Workspace/master`
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/rosmod
Workspace:                   /home/jeb/compilation/VADL+Workspace/master
--------------------------------------------------------------------------------
Source Space:       [exists] /home/jeb/compilation/VADL+Workspace/master/src
Log Space:         [missing] /home/jeb/compilation/VADL+Workspace/master/logs
Build Space:        [exists] /home/jeb/compilation/VADL+Workspace/master/build
Devel Space:        [exists] /home/jeb/compilation/VADL+Workspace/master/devel
Install Space:      [unused] /home/jeb/compilation/VADL+Workspace/master/install
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------------------------
[build] Found '5' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild               
Finished  <<< catkin_tools_prebuild                [ 9.2 seconds ]
Starting  >>> Definitions                         
No handlers could be found for logger "trollius"
_______________________________________________________________________________
Warnings   << Definitions:make /home/jeb/compilation/VADL+Workspace/master/logs/Definitions/build.make.000.log
WARNING: Catkin package name "Definitions" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, and underscores.

WARNING: Catkin package name "Definitions" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, and underscores.

WARNING: Catkin package name "Definitions" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, and underscores.

WARNING: Catkin package name "Definitions" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, and underscores.

WARNING: Catkin package name "Definitions" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, and underscores.

cd /home/jeb/compilation/VADL+Workspace/master/build/Definitions; catkin build --get-env Definitions | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

...............................................................................
Finished  <<< Definitions                          [ 16.7 seconds ]
[build] Summary: All 2 packages succeeded!
[build]   Ignored:   4 packages were skipped or are blacklisted.
[build]   Warnings:  1 packages succeeded with warnings.
[build]   Abandoned: None.
[build]   Failed:    None.
[build] Runtime: 26.0 seconds total.
[build] Note: Workspace packages have changed, please re-source setup files to use them.
cp: cannot stat ‘devel/lib/*.so’: No such file or directory
```

DWatkins, 2017-12-04:
could possibly be an out of date rosmod-comm

finger563, 2017-12-04:
Looks like it might be looking for binaries but only message definitions
we're built. Chalk it up as a bug since it doesn't handle the case when the
only packages that are built do not contain components

William Emfinger, Ph.D.
Chief Technology Officer
MAX Mobility, LLC.
5425 Crossings Blvd, Antioch TN
615-953-5360

commit f85251d9 (2017-12-04): closes #208 only copy bin if there are components built; only copy share if there are types built.
closed 2017-12-04 by finger563