rosmod/webgme-rosmod #210: Start external ros nodes

DWatkins, 2017-12-19:
External nodesshould be run by the `roslaunch <node name> <launchfile>` command.

Ideally these external nodes would be described somehow in the model instead of just being a command line argument to pass in during `Run Experiment`. Would it make sense to specify an external node in the system model since we are going to move system libraries there as well? 

Else is could be specified in the software model and added to deployments the same way that components are?

Relevant attributes for a node component:

```
node name
launch file name
```

A node could possibly contain its own external defintions?

Example usage:

To interface with a 3DConnexion mouse it is easiest to use the existing `spacenav_node` ros package. This is launched using `roslaunch spacenav_node classic.launch`. The spacenav node then publishes the following topics

```
/spacenav/joy
/spacenav/offset
/spacenav/rot_offset
/spacenav/twist
```

http://wiki.ros.org/spacenav_node

DWatkins, 2017-12-19:
currently the topics can be defined in external definitions, but the node has be be launched manually before deployment.

DWatkins, 2017-12-19:
Just like system libraries, the onus is on the user to make sure the package is installed on the system being deployed to.

finger563, 2017-12-19:
add new `External Node` meta-type to deployments which includes a `run command` and possibly a `stop command` which allows the `RunExperiment` and `StopExperiment` plugins to know what to do :)

finger563, 2017-12-27:
@DWatkins what's the best accepted practice for stopping these processes later? just to send them a `SIGKILL / SIGTERM`?

DWatkins, 2017-12-27:
Yep!

finger563, 2017-12-27:
I've update the meta and RunExperiment plugin to support `External Nodes`, I'll leave you to test using them (since you've already got some that you use) to see if the PID is properly captured by the current code.  If so, then I can save the PIDs into the model and then kill them with `StopExperiment`

DWatkins, 2017-12-27:
http://wiki.ros.org/rosnode

`rosnode kill <node name> ` also works. Must be called before roscore is killed.

DWatkins, 2017-12-27:
killing via PID is probably better. rosnode searches for the node name instead of the type or PID. Unfortunately node name is defined within a launch file and is pretty arbitrary.

commit c2d8a1db (2017-12-27): closes #210 Finished support for managing "External Nodes" within an experiment.
closed 2017-12-27 by finger563