rosmod/webgme-rosmod #220: What would be required to generate and export a .webgmexm of a software model based on a deployment

DWatkins, 2018-03-06:
I.E. Generate a software model tree that is a subset of the current project's software model that contains only the relevant components to the deployment at hand. 

# Example
```
Project Software Model: 
  Library-1
  Library-2
  Package A
    Component A.1
      Uses Library-1
    Component A.2
    Component A.3
  Package B
    Component B.1
    Component B.2
      Uses Library-2
```
```
Deployment Model:
  Container
    Node 1
      Component A.1
    Node 2
      Component A.2
       Component B.1
```
```
.webgmexm Software Model:
  Library-1
  Package A
    Component A.1
    Component A.2
  Package B
    Component B.1
```

finger563, 2018-03-07:
realistically we would want the plugin to act on an `Experiment` so that it will find the deployment, system model, and from there the relevant components that need to be exported.

finger563, 2018-05-04:
Looks like we can use `exportModelsToFile` from [webgme-engine/serialization.js](https://github.com/webgme/webgme-engine/blob/master/src/common/util/serialization.js#L101)