rosmod/webgme-rosmod #221: Explore remapping of topics in a deployment

DWatkins, 2018-03-07:
To be truly modular, messages of the same type but different names among various components need to be "connectable". This allows development of specific software components that are application agnostic. 

Example being a controller that publishes an *init* command of type `std_msgs/Empty` that is remapped to connect to an rnet can bus component that subscribes to an *injectErrorFrames* command of type `std_msgs/Empty` 

Otherwise, the case now is that either the controller component must publish *injectErrorFrames* directly, and thus loses generality, or the rnet bus interface subscribes to *init* and its semantics does not match its function.

finger563, 2018-03-07:
need to look into running `rosmod_actor` with `rosrun` or `roslaunch` and see what else would be required to support remapping.

finger563, 2018-03-07:
Then would need to see how we might handle such remapping in the `CommViz`

finger563, 2018-03-07:
Can be accomplished by adding a similar `Arguments` as what's in the `External Node` which can be converted to `<original>:=<new>` command line args.  have tested with existing `rosmod_actor` executable and it appears to work :)

commit 53981ced (2018-03-07): closes #221 added "Arguments" to "Node" for specifying command line arguments for ROSMOD nodes which enables remapping of topics.
closed 2018-03-07 by finger563