rosmod/webgme-rosmod #227: Cannot use rosmsg show custom topic info

DWatkins, 2018-03-19:
Maybe need to add the `src` folder to a path somewhere? I will investigate further.

finger563, 2018-03-23:
currently only way i've gotten it to work is to source the `devel/setup.zsh` that's generated - we may need to copy that over in order to be able to do this...

finger563, 2018-03-23:
it's weird because `roscd` `rostopic echo` and such all work just fine

DWatkins, 2018-03-23:
what is the full value of ROS_ROOT and PYTHONPATH?

DWatkins, 2018-03-28:
one issue is there are no `.msg` files in the the generated `share` folder. Copying a the .msg from the `src` folder doesn't fix the problem, so something else is messing up too.

DWatkins, 2018-03-28:
also need to copy the `cmake` folder from `devel/share/[package name]` to the `share` folder in rosmod's generated folder

`cp -rL` to deference symbolic links

moving cmake folder and the `.msg` file fixes the problem

commit 80703170 (2018-03-28): fixes #227 properly manage install targets for ROSMOD packages to enable use of ros cmd-line tools
closed 2018-03-28 by finger563