rosmod/webgme-rosmod #228: ROSMOD fails to set correct message path when compiling messages

jeholliday, 2018-03-22:
Example Message:
```
std_msgs/Bool pid
geometry_msgs/Twist twist
```

Example error:
```
Could not find messages which
  '/home/vur/compilation/VURC+RMC/masterDrive/src/common/msg/DrivetrainControl.msg'
  depends on.  Did you forget to specify generate_messages(DEPENDENCIES ...)?
  Cannot locate message [Twist]: unknown package [geometry_msgs] on search
  path [{{'std_msgs': ['/opt/ros/kinetic/share/std_msgs/cmake/../msg'],
  'common': ['/home/vur/compilation/VURC+RMC/masterDrive/src/common/msg']}}]
Call Stack (most recent call first):
  /opt/ros/kinetic/share/genmsg/cmake/genmsg-extras.cmake:307 (include)
  CMakeLists.txt:43 (generate_messages)
```

DWatkins, 2018-03-22:
the issue is that `geometry_msgs` is not added to `generate_messages()` in the project's CMakeLists.txt

needs to be added such that `generat_message()` looks like this

```cmake
generate_messages(
 DEPENDENCIES
 std_msgs
 geometry_msgs
)
```

finger563, 2018-03-22:
will fix this shortly (tonight probably) - i'll update it to regex out `${package_name}/${msg_name}` and add all `${package_name}`s it finds to the dependencies

commit 0561edbb (2018-03-23): closes #228 parse message / service definitions for dependent types and properly configure build system accordingly.
closed 2018-03-23 by finger563