rosmod/webgme-rosmod #243: Support for Deployment Visualizing External Messages

jeholliday, 2018-03-31:
Right now, the deployment commsviz only shows the message connections between self defined messages. It would be good if it could also show connections between external messages and external nodes.

finger563, 2018-04-01:
Unfortunately we have no model for what topics external nodes pub/sub/provide/require and on what names - we've talked about adding this but there is no real standard way of discovering them - so we'd have to have the user manually make them within the model (which isn't great because it is error prone and increases the development burden for something that's purely viz).  That being said, we're still considering it.

finger563, 2018-04-02:
@jeholliday we're definitely open to suggestions for how you might want to represent / model such topics / advertisements

jeholliday, 2018-04-04:
I understand how visualizing messages coming from external nodes would be difficult. Having the external nodes in the visualization isn't necessary but external messages would be useful. It appears that even if a ROSMOD defined node is publishing an external message, and then another ROSMOD defined node subscribes to it, that message doesn't show up in the visualization.

finger563, 2018-04-04:
Yea @dwatkins and @jeholliday that's an easy thing to add - open a separate issue for it and I'll take care of it tomorrow