rosmod/webgme-rosmod #253: Errors in auto-generated code

jeholliday, 2018-04-13:
I've just updated to the latest version, and I'm now receiving errors when I try to compile that appear to be coming from the auto-generated code. Here are the errors:

```
Errors     << common:make /home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/logs/common/build.make.001.log                                                                                                              
In file included from /home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp:7:0:
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/include/common/JoystickDrivetrainController.hpp:60:8: error: ‘void JoystickDrivetrainController::init_timer_operation(const ros::TimerEvent&)’ marked ‘override’, but does not override
   void init_timer_operation(const ros::TimerEvent& event) override;
        ^
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp: In member function ‘void JoystickDrivetrainController::init_timer_operation(const ros::TimerEvent&)’:
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp:49:3: error: ‘trace’ was not declared in this scope
   trace->raw_log("init_timer_operation, %d.%09d, %d.%09d, %d.%09d, %d.%09d, %d.%09d",
   ^
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp: In member function ‘void JoystickDrivetrainController::JoystickSubscriber_operation(const ros::MessageEvent<const sensor_msgs::Joy_<std::allocator<void> > >&)’:
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp:103:3: error: ‘trace’ was not declared in this scope
   trace->raw_log("JoystickSubscriber, %d.%09d, %d.%09d, %d.%09d, %d.%09d, %d.%09d",
   ^
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp: In member function ‘virtual void JoystickDrivetrainController::startUp()’:
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp:150:13: error: ‘class Logger’ has no member named ‘enable_logging’
     logger->enable_logging();
             ^
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp:161:5: error: ‘trace’ was not declared in this scope
     trace->enable_logging();
     ^
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp:180:29: error: ‘nh_’ was not declared in this scope
   this->JoystickPublisher = nh_.advertise<common::DrivetrainControl>(
                             ^
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp:180:68: error: expected primary-expression before ‘>’ token
   this->JoystickPublisher = nh_.advertise<common::DrivetrainControl>(
                                                                    ^
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp:194:19: error: no matching function for call to ‘ros::SubscribeOptions::create(const char*, int, boost::_bi::bind_t<void, boost::_mfi::mf1<void, JoystickDrivetrainController, const ros::MessageEvent<const sensor_msgs::Joy_<std::allocator<void> > >&>, boost::_bi::list2<boost::_bi::value<JoystickDrivetrainController*>, boost::arg<1> > >, ros::VoidPtr, rosmod::CallbackQueue*)’
        &comp_queue);
                   ^
In file included from /opt/ros/kinetic/include/ros/node_handle.h:42:0,
                 from /opt/ros/kinetic/include/ros/ros.h:45,
                 from /opt/ros/kinetic/include/actionlib/client/simple_action_client.h:45,
                 from /home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/include/common/JoystickDrivetrainController.hpp:11,
                 from /home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp:7:
/opt/ros/kinetic/include/ros/subscribe_options.h:154:27: note: candidate: template<class M> static ros::SubscribeOptions ros::SubscribeOptions::create(const string&, uint32_t, const boost::function<void(const boost::shared_ptr<const M>&)>&, const VoidConstPtr&, ros::CallbackQueueInterface*)
   static SubscribeOptions create(const std::string& topic, uint32_t queue_size,
                           ^
/opt/ros/kinetic/include/ros/subscribe_options.h:154:27: note:   template argument deduction/substitution failed:
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp:194:19: note:   cannot convert ‘&((JoystickDrivetrainController*)this)->JoystickDrivetrainController::<anonymous>.Component::comp_queue’ (type ‘rosmod::CallbackQueue*’) to type ‘ros::CallbackQueueInterface*’
        &comp_queue);
                   ^
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp:204:11: error: no matching function for call to ‘ros::TimerOptions::TimerOptions(ros::Duration, boost::_bi::bind_t<void, boost::_mfi::mf1<void, JoystickDrivetrainController, const ros::TimerEvent&>, boost::_bi::list2<boost::_bi::value<JoystickDrivetrainController*>, boost::arg<1> > >, rosmod::CallbackQueue*, bool, bool)’
      false); 
           ^
In file included from /opt/ros/kinetic/include/ros/timer.h:33:0,
                 from /opt/ros/kinetic/include/ros/node_handle.h:36,
                 from /opt/ros/kinetic/include/ros/ros.h:45,
                 from /opt/ros/kinetic/include/actionlib/client/simple_action_client.h:45,
                 from /home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/include/common/JoystickDrivetrainController.hpp:11,
                 from /home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/JoystickDrivetrainController.cpp:7:
/opt/ros/kinetic/include/ros/timer_options.h:53:3: note: candidate: ros::TimerOptions::TimerOptions(ros::Duration, const TimerCallback&, ros::CallbackQueueInterface*, bool, bool)
   TimerOptions(Duration _period, const TimerCallback& _callback, 
   ^
/opt/ros/kinetic/include/ros/timer_options.h:53:3: note:   no known conversion for argument 3 from ‘rosmod::CallbackQueue*’ to ‘ros::CallbackQueueInterface*’
/opt/ros/kinetic/include/ros/timer_options.h:42:3: note: candidate: ros::TimerOptions::TimerOptions()
   TimerOptions()
   ^
/opt/ros/kinetic/include/ros/timer_options.h:42:3: note:   candidate expects 0 arguments, 5 provided
/opt/ros/kinetic/include/ros/timer_options.h:40:20: note: candidate: ros::TimerOptions::TimerOptions(const ros::TimerOptions&)
 struct ROSCPP_DECL TimerOptions
                    ^
/opt/ros/kinetic/include/ros/timer_options.h:40:20: note:   candidate expects 1 argument, 5 provided
/opt/ros/kinetic/include/ros/timer_options.h:40:20: note: candidate: ros::TimerOptions::TimerOptions(ros::TimerOptions&&)
/opt/ros/kinetic/include/ros/timer_options.h:40:20: note:   candidate expects 1 argument, 5 provided
In file included from /home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/src/common/Drivetrain.cpp:7:0:
/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/include/common/Drivetrain.hpp:63:8: error: ‘void Drivetrain::init_timer_operation(const ros::TimerEvent&)’ marked ‘override’, but does not override
   void init_timer_operation(const ros::TimerEvent& event) override;
[... 105 lines cut ...]
/opt/ros/kinetic/include/ros/timer_options.h:40:20: note:   candidate expects 1 argument, 5 provided
make[2]: *** [CMakeFiles/JoystickDrivetrainController.dir/src/common/JoystickDrivetrainController.cpp.o] Error 1
make[1]: *** [CMakeFiles/JoystickDrivetrainController.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Drivetrain.dir/src/common/Drivetrain.cpp.o] Error 1
make[1]: *** [CMakeFiles/Drivetrain.dir/all] Error 2
make: *** [all] Error 2
```

Additionally, I'm receiving an error about the search path:
```
Warnings   << common:cmake /home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/logs/common/build.cmake.001.log                                                                                                          [... cut ...]
  Cannot generate a safe runtime search path for target Drivetrain because
  there is a cycle in the constraint graph:

    dir 0 is [/opt/ros/kinetic/lib]
    dir 1 is [/opt/rosmod/lib]
    dir 2 is [/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/install/lib]
      dir 3 must precede it due to runtime library [libpid.so]
    dir 3 is [/home/vur/webgme-rosmod/generated/VURC+RMC/driveTest2/TestBoard/src/common/../../devel/lib]
      dir 2 must precede it due to runtime library [libpid.so]

  Some of these libraries may not be found correctly.
```

finger563, 2018-04-13:
hey @jeholliday that's from a recent pull you did - the latest code has been updated to generate for a new install target that's no longer `rosmod-comm` but rather [rosmod-actor](https://github.com/rosmod/rosmod-actor) - see the latest version of the webgme rosmod readme specifically you should install

* [rosmod-actor](https://github.com/rosmod/rosmod-actor)
* and (optionally - if you want support for actions) [actionlib](https://github.com/rosmod/actionlib)

finger563, 2018-04-13:
The updated readme has instructions for installing actionlib, and rosmod-actor has similar install instructions as rosmod-comm **NOTE: this will have to be done for each target**

jeholliday, 2018-04-13:
This successfully resolved my issue. Anyone else who has this problem should note that by default the new rosmod actor gets installed to /opt/rosmod-actor. This required me to update the workspace location of each host in systems.

closed 2018-04-13 by jeholliday